xpack-dev-tools/qemu #7: -daemonize failes with "operation not permitted"

LnnrtS, 2020-12-19:
### Description

emulator does not start with `-daemonize` option

### Steps to Reproduce

works:
```
$ qemu-system-gnuarmeclipse -board STM32F4-Discovery -mcu STM32F407VG -image some/elf/file -S -nographic
QEMU 2.8.0-9 monitor - type 'help' for more information
(qemu)
```

fails:
```
$ qemu-system-gnuarmeclipse -board STM32F4-Discovery -mcu STM32F407VG -image some/elf/file -S -nographic -daemonize
qemu: qemu_mutex_unlock: Operation not permitted
```

### Versions

`2.8.0-9.2`

### Additional information

A quick search bings up [this patch](https://patchwork.kernel.org/project/qemu-devel/patch/1498234919-27316-13-git-send-email-kwolf@redhat.com/)

ilg-ul, 2020-12-19:
I tried the suggested patch, but it did not help.

Any other ideas?

Fasten90, 2022-02-21:
I get this issue at docker container, but I don't at other situation. Is it help for you somehow?

ilg-ul, 2022-02-21:
I suggest you try qemu-system-arm.