xpack-dev-tools/qemu #15: rom: requested regions overlap

LnnrtS, 2021-10-17:
### Description

I am running in the exactly the same bug as described [here](https://github.com/beckus/qemu_stm32/issues/27).  
In one version of the linker script it works in qemu and not on the board and in the other version it is the other way around.

Explanation and possible fix described in the [last comment](https://github.com/beckus/qemu_stm32/issues/27#issuecomment-595696040)

### Versions

* qemu 2.8.0-12.1
* Debian Stretch

ilg-ul, 2021-10-17:
I transferred the issue from qemu-xpack.git to qemu.git, since it is related to QEMU functionality, not the build scripts..

ilg-ul, 2021-10-19:
The two patches were applied and v2.8.0-13 released.

@LnnrtS, could you check if these patches solved your issue?